genestealer/everblu-meters-esp8266-improved #4: WiFi connectivity issues - possible solution

fraserp, 2025-03-29:
I've been having problems with this build in getting it to connect reliably to my WiFi. TBH that's not entirely unusual for IoT devices on the whole...

The problem seems to be how the WiFi stack is initiated in this build... but I can't see where its doing this in the code (usually its WiFi.Begin(..) but that's not there?)... and my router isn't happy with the connection (possibly non compliant or missing config/flags?) and repeatedly rejects it 

I can get around this by forcing it to 802.11g using WiFi.setPhyMode(WIFI_PHY_MODE_11G) in the void setup() code at the bottom of everblu-meters-esp8266.cpp. So far this seems to have solved it, but I'm going to watch it for a few days to be sure. If that works I will submit a pull request.

More recent ASUS routers seem to be notoriously picky about the connection. Not only do I have an Asus router (RT-AX88U) but I also run modified firmware on it (Merlin) so I am likely also making my own life harder 😂.... To be fair my previous Asus router (RT-AC86U) was more forgiving.

genestealer, 2025-04-11:
@fraserp Could it be that your WiFi access point has 5 GHz band steering enabled? Some routers (especially ASUS models) try to automatically steer devices to the 5 GHz band, which the ESP8266 doesn't support. This can cause unreliable connection attempts or outright failures.

If you haven't already, try either:
- Disabling band steering
- Splitting your SSIDs for 2.4 GHz and 5 GHz into separate names, then connect the ESP8266 only to the 2.4 GHz one

Might be worth checking before digging too deep into PHY mode settings.

fraserp, 2025-04-13:
I have it set up as split bands - so 2.4 and 5 GHz are separate SSIDs already. I tried a number of things to resolve this and this was the only thing I found to work.

genestealer, 2025-04-13:
@fraserp 
I have implemented the requested change to address the WiFi connectivity issue. 

Set value in `ENABLE_WIFI_PHY_MODE_11G` to `0` in `private.h`

This allows for easier control over the Wi-Fi PHY mode. Let me know if further adjustments are nee

closed 2025-04-13 by genestealer

fraserp, 2025-04-13:
Great stuff thanks!