genestealer/everblu-meters-esp8266-improved #5: What about 868MHz

llaumgui, 2025-04-12:
Hi,

I have a 868MhZ version. Is it planned to adapt your project ?

genestealer, 2025-04-13:
@llaumgui 
I have created a branch to include support for 868 MHz frequency and improved frequency discovery instructions.

Please test it and let me know.

https://github.com/genestealer/everblu-meters-esp8266-improved/tree/868mhz-test

llaumgui, 2025-04-23:
OK, my water meter seem to be crypted. Your module can uncrypt ?

llaumgui, 2025-04-23:
When I scan, I have a lot of: 
"␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x�␀�x␀xxx␀�x"

For information, I have a ESP32 with a C11001, but a 868 version.

genestealer, 2025-04-24:
@llaumgui 
Thank you for the information. 

Could you confirm the model of water metre transceiver? If you upload a photograph that would be super helpful. You may wish hide the serial number and barcode. 

From looking at ha-watermetre and reading through its code it appears only to be compatible with water metres that are using the wM-Bus protocol.

The software in this repository is for meters using the Cyble RADIAN protocol. Which is different to the wM-BUS protocol.

I would need to know further information on the exact make and model of water metre transceiver you have. A photo would really help.

llaumgui, 2025-04-30:
I just receive my D1 mini. I have build but I have exception on scan:

```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v0005e060
~ld
��␃�␛�o�r��o|�␄l␌�␌l ␃␜c�␛␂␄�|␂r�␂l�o�␄�o�␀␌$`␂��{�d�$␒�␄␌␌�

Everblu Meters ESP8266 Starting...
Water usage data for Home Assistant
https://github.com/genestealer/everblu-meters-esp8266-improved
Target meter: 25-02244363
Target frequency: 868.0000000000

###### FREQUENCY DISCOVERY ENABLED (868 MHz) ######
Starting Frequency Scan...
Test frequency : 868.000000

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

Exception (4):
epc1=0x402092fe epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffff7a0 end: 3fffffd0 offset: 0160
3ffff900:  00000000 3ffff951 00000001 4020952c  
3ffff910:  0007a120 00000000 0000000f 000003e8  
3ffff920:  00000002 3ffff950 3fff1fc8 40201065  
3ffff930:  00000002 3fff0001 3fff1fc8 4020106f  
3ffff940:  3fff1b4c 00000096 00000064 40201124  
3ffff950:  3fff00ff 00000096 00000064 402010e0  
3ffff960:  0000ffff 3fff1b4c 3fff1fc8 402014f7  
3ffff970:  00000000 00000001 e0000000 402016a2  
3ffff980:  00000064 3ffff9b0 00000000 402010e0  
3ffff990:  0000ffff 3fff1b4c 0000000a 0000004c  
3ffff9a0:  3ffffe60 3ffffef0 00000000 40201993  
3ffff9b0:  feefeffe feefeffe feefeffe feefeffe  
3ffff9c0:  feefeffe feefeffe feefeffe feefeffe  
3ffff9d0:  feefeffe feefeffe feefeffe feefeffe  
3ffff9e0:  feefeffe feefeffe feefeffe feefeffe  
3ffff9f0:  feefeffe feefeffe feefeffe feefeffe  
3ffffa00:  feefeffe feefeffe feefeffe feefeffe  
3ffffa10:  feefeffe feefeffe feefeffe feefeffe  
3ffffa20:  feefeffe feefeffe feefeffe feefeffe  
3ffffa30:  feefeffe feefeffe feefeffe feefeffe  
3ffffa40:  feefeffe feefeffe feefeffe feefeffe  
3ffffa50:  feefeffe feefeffe feefeffe feefeffe  
3ffffa60:  feefeffe feefeffe feefeffe feefeffe  
3ffffa70:  feefeffe feefeffe feefeffe feefeffe  
3ffffa80:  feefeffe feefeffe feefeffe feefeffe  
3ffffa90:  feefeffe feefeffe feefeffe feefeffe  
3ffffaa0:  feefeffe feefeffe feefeffe feefeffe  
3ffffab0:  feefeffe feefeffe feefeffe feefeffe  
3ffffac0:  feefeffe feefeffe feefeffe feefeffe  
3ffffad0:  feefeffe feefeffe feefeffe feefeffe  
3ffffae0:  feefeffe feefeffe feefeffe 4021e9b5  
[... 80 lines cut ...]
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v0005e060
~ld
��␃�␒�n��{��n|�␄l␄�␌$ ␂␜b���␀␌�|␂{�␂l�o�␌�n�␀␄l ␃��r�l�l␒�␌␄␌�

Everblu Meters ESP8266 Starting...
Water usage data for Home Assistant
https://github.com/genestealer/everblu-meters-esp8266-improved
Target meter: 25-02244363
Target frequency: 868.0000000000

###### FREQUENCY DISCOVERY ENABLED (868 MHz) ######
Starting Frequency Scan...
Test frequency : 868.000000
```

llaumgui, 2025-04-30:
OK, my fault, `GDO0` is `1` for a Weemos.

Now my monitor display:

```
Everblu Meters ESP8266 Starting...
Water usage data for Home Assistant
https://github.com/genestealer/everblu-meters-esp8266-improved
Target meter: 25-02244363
Target frequen�
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v0005e070
~ld
��␂�␛�o�r��n|�␄�$�␜{l␜c�␛␃␄�|␂r�␂d�o�␄�o�␀␌$`␂���N�l�␜�␛�␄␌␄�
```

genestealer, 2025-04-30:
It looks like it's rebooting to me. Are you sure you've modified the platform IO device configuration file to tell it you're using a D1 mini?

genestealer, 2025-04-30:
Please ensure you have set the `platformio.ini` file for you board.

See the latest https://github.com/genestealer/everblu-meters-esp8266-improved/blob/main/platformio.ini

llaumgui, 2025-05-01:
Hi,

I have a:
 * [D1 mini pro CP2104 4MB](https://fr.aliexpress.com/item/1005007475726143.html?spm=a2g0o.order_list.order_list_main.23.76c65e5bcd1KRH&gatewayAdapt=glo2fra)
 * [C11001 868MHz](https://fr.aliexpress.com/item/1005007129880189.html?spm=a2g0o.order_list.order_list_main.11.76c65e5bcd1KRH&gatewayAdapt=glo2fra)

My platformio.ini:

```
[env:d1_mini_pro]
platform = espressif8266
board = d1_mini_pro
framework = arduino

; Upload and monitor options
upload_speed = 460800
monitor_speed = 115200

; Library options
lib_deps =
    plapointe6/EspMQTTClient  ; MQTT client library for ESP8266/ESP32
    Arduino                   ; Core Arduino library for ESP8266
    ESP8266WiFi               ; Wi-Fi library for ESP8266
    ESP8266mDNS               ; mDNS library for ESP8266
    ArduinoOTA                ; OTA (Over-the-Air) update library for ESP8266
```

My private.h:

```
// WiFi credentials
#define secret_wifi_ssid "xxx"          // WiFi SSID (Network Name)
#define secret_wifi_password "xxx"  // WiFi Password

// MQTT server configuration
#define secret_mqtt_server "192.168.xx.xx" // MQTT Server IP Address
#define secret_clientName "everbluMeters"   // MQTT Client Name

// MQTT authentication
#define secret_mqtt_username "watermeter"        // MQTT Username
#define secret_mqtt_password "xxx"  // MQTT Password

// NTP server for time synchronization
#define secret_local_timeclock_server "0.fr.pool.ntp.org" // NTP Server Address

// Meter-specific configuration
#define METER_YEAR 25        // Last two digits of the year printed on the meter (e.g., 2019 is 19)
#define METER_SERIAL 2244xxx  // Meter Serial Number (omit leading zero)
#define FREQUENCY 868.950000 // Frequency of the meter (discovered via test code)
#define GDO0 1               // Pin on ESP8266 used for GDO0 (General Digital Output 0)
```

llaumgui, 2025-05-03:
@genestealer > My GD0 definition is good ? Is there others mapping to do ?

genestealer, 2025-05-13:
I'm sorry, as far as I can see it will not work with the Cyble 5

llaumgui, 2025-05-13:
Thx.

My GDO0 and mapping from your repo is good for D1 ?

genestealer, 2025-05-13:
Yes for the Itron EverBlu Cyble Enhanced. 
Its not tested with the Cyble 5.