genestealer/everblu-meters-esp8266-improved #6: missing SConscript file 'build.py' - PlatformIO - WEMOS D1 min pro

ChristophZIE, 2025-04-28:
Hello, great work,
When compiling, I receive an error missing SConscript file 'build.py'  in PlatformIO, unfortunely I'm a newbie to platformio and I do not find a solution on the web. 
I tried  by project wizard to parameter the right board, but no succes.
Did you try a Wemos ?

Thanks a lot for your help in advance.

Best regards 
Chris

genestealer, 2025-04-30:
Just tested building it for a WEMOS D1 min pro and it built fine.

[image]

I added a more detailed `platformio.ini` file to the latest version https://github.com/genestealer/everblu-meters-esp8266-improved/commit/fc966d4c0382a5ee7d17fa7e1c21f4ee457ec7ba

Please check your `platformio.ini` looks like this https://github.com/genestealer/everblu-meters-esp8266-improved/blob/main/platformio.ini

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

; Adjust as needed. Make sure you only have one env in this config file.
; See:
;  https://docs.platformio.org/en/stable/boards/index.html#espressif-8266
;  https://docs.platformio.org/en/stable/boards/index.html#espressif-32
;  and so on for other boards.

; ####################################################################
; EXAMPLES
; ####################################################################

; Uncomment the following line to use the WeMos D1 mini Pro board. https://docs.platformio.org/en/stable/boards/espressif8266/d1_mini_pro.html
[env:d1_mini_pro]
platform = espressif8266
board = d1_mini_pro
framework = arduino

; Uncomment the following line to use the nodemcuv2 ESP8266 board. https://docs.platformio.org/en/stable/boards/espressif8266/nodemcuv2.html
; [env:nodemcuv2]
; platform = espressif8266
; board = nodemcuv2
; framework = arduino

; Uncomment the following 4 lines to use the Adafruit HUZZAH ESP8266 board. https://docs.platformio.org/en/stable/boards/espressif8266/huzzah.html
; [env:huzzah]
; platform = espressif8266
; board = huzzah
; framework = arduino

; ####################################################################

; Upload and monitor options
upload_speed = 460800
monitor_speed = 115200

; Library options
lib_deps =
    plapointe6/EspMQTTClient  ; MQTT client library for ESP8266/ESP32
    Arduino                   ; Core Arduino library for ESP8266
    ESP8266WiFi               ; Wi-Fi library for ESP8266
    ESP8266mDNS               ; mDNS library for ESP8266
    ArduinoOTA                ; OTA (Over-the-Air) update library for ESP8266

; ####################################################################
; Over The Air (OTA) upload options
; ####################################################################
; Uncomment the lines below to upload via ethernet. Replace the IP address with your ESP8266's IP address. Re-comment the lines out if you want to upload via USB.
; upload_protocol = espota
; upload_port = 192.168.2.21
; upload_port = /dev/ttyUSB0 ; or this on linux
```

genestealer, 2025-05-13:
Close as assumed fixed with new update linked above.

closed 2025-05-13 by genestealer